EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #178: Getting list index out of range error when fetching boxes from model

aansrajpoot, 2024-03-05:
I trained the yolo8 nano model from Ultrlytics and downloaded the .tflite model for inferencing. Now when I am running the code I am getting this error. I don't know how to solve this issue. So, that I can get boxes, classes, and scores. This is how I'm importing Interpreter.
Import tensorflow as tf
interpreter = tf.lite.Interpreter(model_path=modelpath)

This is the error I'm getting:
Traceback (most recent call last):
  File "E:\PycharmProjects\tflitemodel\modeltest2.py", line 130, in <module>
    tflite_detect_images(PATH_TO_MODEL, PATH_TO_IMAGES, PATH_TO_LABELS, min_conf_threshold, images_to_test)
  File "E:\PycharmProjects\tflitemodel\modeltest2.py", line 68, in tflite_detect_images
    boxes = interpreter.get_tensor(output_details[1]['index'])[0] # Bounding box coordinates of detected objects
                                   ~~~~~~~~~~~~~~^^^
IndexError: list index out of range

Please help me to solve this issue.
Thanks

4tabierg, 2024-05-20:
I have same issue. Did you find any solution for it?

EdjeElectronics, 2024-05-20:
Sorry guys, this code doesn't support YOLOv8 models, even if they've been converted to .tflite format. You have to add some extra post-processing of the results. You can find more info on the Ultralytics repository. Good luck!

closed 2024-05-20 by EdjeElectronics