EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #180: using object detection with TPU has some error(Segmentation fault).

sangyo1, 2024-03-21:
I was watching the youtube video and run the detection model on my rpasberry pi with TPU.
https://www.youtube.com/watch?v=qJMwNHQNOVU&ab_channel=EdjeElectronics

I was able to run without TPU, it ran with 8FPS with my webcam
```
python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model
```
However, when I ran with TPU
```
python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model --edgetpu
```
It kept gave me "Segmentation fault" message.
What is wrong and how do I fix it?

SamuelKollar, 2024-03-23:
Hey, I get the same error did you find any fix for it?

SamuelKollar, 2024-03-24:
Resolved this issue, I just installed the Edge TPU runtime library from https://github.com/feranick/libedgetpu/releases/tag/v16.0TF2.13.1-2 for the correct version of my RaspberryPi OS and it resolved the issue.

sangyo1, 2024-03-25:
> Resolved this issue, I just installed the Edge TPU runtime library from https://github.com/feranick/libedgetpu/releases/tag/v16.0TF2.13.1-2 for the correct version of my RaspberryPi OS and it resolved the issue.

Could you tell me more specific?
[libedgetpu-dev_16.0tf2.13.1-2.bookworm_arm64.deb](https://github.com/feranick/libedgetpu/releases/download/v16.0TF2.13.1-2/libedgetpu-dev_16.0tf2.13.1-2.bookworm_arm64.deb)
I download above file in my rapsberry pi, and ```sudo dpkg -i ``` the package.
I ran ```python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model --edgetpu```
I still get the same error message.
Do you have any link that how to solve this problem?

SamuelKollar, 2024-03-26:
I specifically installed libedgetpu1-std_16.0tf2.13.1-2.ubuntu22.04_arm64.deb form the link above, as it is the standard runtime and it is specific for the OS that is running on my pi4, I also installed python 3.11 specific wheel from https://pypi.org/project/tensorflow/2.12.0/#files. 

After that I just ran the command for TFLite_detection_webcam.py and it ran with no issues.

closed 2024-03-29 by sangyo1