EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #193: ModuleNotFoundError: No module named 'object_detection'

svenancio, 2025-02-02:
Hello. First of all, thank you so much for sharing this code with us.

I'm running this 5th cell in Colab

```
# Run Model Bulider Test file, just to verify everything's working properly
!python /content/models/research/object_detection/builders/model_builder_tf2_test.py
```

and getting the following error

```
2025-02-02 22:43:06.049200: I tensorflow/core/util/port.cc:153] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable TF_ENABLE_ONEDNN_OPTS=0.
2025-02-02 22:43:06.066832: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:477] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
E0000 00:00:1738536186.087456    7358 cuda_dnn.cc:8310] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
E0000 00:00:1738536186.093633    7358 cuda_blas.cc:1418] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2025-02-02 22:43:06.114862: I tensorflow/core/platform/cpu_feature_guard.cc:210] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
Traceback (most recent call last):
  File "/content/models/research/object_detection/builders/model_builder_tf2_test.py", line 24, in <module>
    from object_detection.builders import model_builder
ModuleNotFoundError: No module named 'object_detection'
```

Tried to find a solution, but no success so far. Previous cells run correctly, despite the warnings on the 4th cell. It seems like an error inside Tensorflow, since this code of yours ran smoothly a few weeks ago.

I'd be glad if you have any tips. Thanks in advance.

sayedidrees, 2025-02-07:
@svenancio This Colab files seems to be outdated.
@EdjeElectronics  Can you please help on this one?

EdjeElectronics, 2025-02-11:
Hey all, I am working on it. It's occurring because Colab recently upgraded from Python 3.10 to Python 3.11. Hopefully will having a solution within the week!

XeroDays, 2025-02-12:
@EdjeElectronics  Great, Looking forward. 👁️👁️

EdjeElectronics, 2025-02-13:
Hey all, a fix has been added. For now, we will just use the fallback runtime version that still has Python 3.10. This fix will probably only be valid for the next couple of weeks.

After initially connecting to the notebook, go to Tools -> Command Palette, search for "runtime", and select "Use fallback runtime version". Then run through the notebook as normal. Full instructions are given in the notebook.

closed 2025-02-13 by EdjeElectronics

SugengGod, 2025-03-08:
I have clicked on fallback runtime version, but when I check the python version it is still 3.11.11

[image]